huaweiwx/STM32GENERIC #10: hello_world error

VictorTagayun, 2018-11-12:
```
sketch\BASIC_HelloWorld.c.o: In function `MainTask':
sketch/BASIC_HelloWorld.c:36: undefined reference to `GUI_Clear'
sketch/BASIC_HelloWorld.c:37: undefined reference to `GUI_SetFont'
sketch/BASIC_HelloWorld.c:38: undefined reference to `LCD_GetXSize'
sketch/BASIC_HelloWorld.c:38: undefined reference to `LCD_GetYSize'
sketch/BASIC_HelloWorld.c:38: undefined reference to `GUI_DispStringAt'
sketch/BASIC_HelloWorld.c:38: undefined reference to `GUI_Font20_1'
sketch\hello_world.ino.cpp.o: In function `STemWin::Init()':
D:\Hobby\Arduino\arduino-1.8.7\hardware\STM32GENERIC_huaweiwx\STM32\libraries\BSP_f429i_discovery\src/STemWin.h:17: undefined reference to `GUI_Init'
sketch\hello_world.ino.cpp.o: In function `setup':
D:\Hobby\Arduino\arduino-1.8.7\hardware\STM32GENERIC_huaweiwx\STM32\libraries\BSP_f429i_discovery\examples\STemWin\hello_world/hello_world.ino:16: undefined reference to `WM_SetCreateFlags'
libraries\BSP_f429i_discovery\STemWin_Config\LCDConf_stm32f429i_disco_MB1075.c.o: In function `HAL_LTDC_LineEventCallback':
D:\Hobby\Arduino\arduino-1.8.7\hardware\STM32GENERIC_huaweiwx\STM32\libraries\BSP_f429i_discovery\src\STemWin_Config/LCDConf_stm32f429i_disco_MB1075.c:367: undefined reference to `GUI_MULTIBUF_ConfirmEx'
collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Discovery F429ZI(180M).

```

huaweiwx, 2018-11-12:
to link the external library STemWIN.lib

VictorTagayun, 2018-11-12:
Worked, thanks!

closed 2018-11-12 by VictorTagayun