huaweiwx/STM32GENERIC #9: draw_2d.ino using Discovery F429ZI

VictorTagayun, 2018-11-12:
```
sketch\draw_2d.ino.cpp.o: In function `STemWin::Init()':
D:\Hobby\Arduino\arduino-1.8.7\hardware\STM32GENERIC_huaweiwx\STM32\libraries\BSP_f429i_discovery\src/STemWin.h:17: undefined reference to `GUI_Init'
sketch\draw_2d.ino.cpp.o: In function `emWinGUI::EnableAlpha(unsigned int)':
D:\Hobby\Arduino\arduino-1.8.7\hardware\STM32GENERIC_huaweiwx\STM32\libraries\STemWin532\src/STemWin532.h:211: undefined reference to `GUI_EnableAlpha'
sketch\draw_2d.ino.cpp.o: In function `emWinGUI::SetBkColor(unsigned long)':
D:\Hobby\Arduino\arduino-1.8.7\hardware\STM32GENERIC_huaweiwx\STM32\libraries\STemWin532\src/STemWin532.h:45: undefined reference to `GUI_SetBkColor'
sketch\draw_2d.ino.cpp.o: In function `emWinGUI::Clear()':
D:\Hobby\Arduino\arduino-1.8.7\hardware\STM32GENERIC_huaweiwx\STM32\libraries\STemWin532\src/STemWin532.h:74: undefined reference to `GUI_Clear'
sketch\draw_2d.ino.cpp.o: In function `emWinGUI::SetTextMode(int)':
D:\Hobby\Arduino\arduino-1.8.7\hardware\STM32GENERIC_huaweiwx\STM32\libraries\STemWin532\src/STemWin532.h:166: undefined reference to `GUI_SetTextMode'
sketch\draw_2d.ino.cpp.o: In function `emWinGUI::SetColor(unsigned long)':
D:\Hobby\Arduino\arduino-1.8.7\hardware\STM32GENERIC_huaweiwx\STM32\libraries\STemWin532\src/STemWin532.h:46: undefined reference to `GUI_SetColor'
sketch\draw_2d.ino.cpp.o: In function `emWinGUI::SetFont(GUI_FONT const*)':
D:\Hobby\Arduino\arduino-1.8.7\hardware\STM32GENERIC_huaweiwx\STM32\libraries\STemWin532\src/STemWin532.h:169: undefined reference to `GUI_SetFont'
sketch\draw_2d.ino.cpp.o: In function `emWinGUI::DispStringHCenterAt(char const*, int, int)':
D:\Hobby\Arduino\arduino-1.8.7\hardware\STM32GENERIC_huaweiwx\STM32\libraries\STemWin532\src/STemWin532.h:136: undefined reference to `GUI_DispStringHCenterAt'
sketch\draw_2d.ino.cpp.o: In function `emWinGUI::SetAlpha(unsigned char)':
D:\Hobby\Arduino\arduino-1.8.7\hardware\STM32GENERIC_huaweiwx\STM32\libraries\STemWin532\src/STemWin532.h:213: undefined reference to `GUI_SetAlpha'
sketch\draw_2d.ino.cpp.o: In function `emWinGUI::SetColor(unsigned long)':
D:\Hobby\Arduino\arduino-1.8.7\hardware\STM32GENERIC_huaweiwx\STM32\libraries\STemWin532\src/STemWin532.h:46: undefined reference to `GUI_SetColor'
sketch\draw_2d.ino.cpp.o: In function `emWinGUI::FillRect(int, int, int, int)':
D:\Hobby\Arduino\arduino-1.8.7\hardware\STM32GENERIC_huaweiwx\STM32\libraries\STemWin532\src/STemWin532.h:113: undefined reference to `GUI_FillRect'
sketch\draw_2d.ino.cpp.o: In function `emWinGUI::SetAlpha(unsigned char)':
D:\Hobby\Arduino\arduino-1.8.7\hardware\STM32GENERIC_huaweiwx\STM32\libraries\STemWin532\src/STemWin532.h:213: undefined reference to `GUI_SetAlpha'
sketch\draw_2d.ino.cpp.o: In function `emWinGUI::SetColor(unsigned long)':
D:\Hobby\Arduino\arduino-1.8.7\hardware\STM32GENERIC_huaweiwx\STM32\libraries\STemWin532\src/STemWin532.h:46: undefined reference to `GUI_SetColor'
sketch\draw_2d.ino.cpp.o: In function `emWinGUI::FillRect(int, int, int, int)':
D:\Hobby\Arduino\arduino-1.8.7\hardware\STM32GENERIC_huaweiwx\STM32\libraries\STemWin532\src/STemWin532.h:113: undefined reference to `GUI_FillRect'
sketch\draw_2d.ino.cpp.o: In function `emWinGUI::SetAlpha(unsigned char)':
D:\Hobby\Arduino\arduino-1.8.7\hardware\STM32GENERIC_huaweiwx\STM32\libraries\STemWin532\src/STemWin532.h:213: undefined reference to `GUI_SetAlpha'
sketch\draw_2d.ino.cpp.o: In function `emWinGUI::SetColor(unsigned long)':
D:\Hobby\Arduino\arduino-1.8.7\hardware\STM32GENERIC_huaweiwx\STM32\libraries\STemWin532\src/STemWin532.h:46: undefined reference to `GUI_SetColor'
sketch\draw_2d.ino.cpp.o: In function `emWinGUI::FillRect(int, int, int, int)':
D:\Hobby\Arduino\arduino-1.8.7\hardware\STM32GENERIC_huaweiwx\STM32\libraries\STemWin532\src/STemWin532.h:113: undefined reference to `GUI_FillRect'
sketch\draw_2d.ino.cpp.o: In function `emWinGUI::SetAlpha(unsigned char)':
D:\Hobby\Arduino\arduino-1.8.7\hardware\STM32GENERIC_huaweiwx\STM32\libraries\STemWin532\src/STemWin532.h:213: undefined reference to `GUI_SetAlpha'
sketch\draw_2d.ino.cpp.o: In function `emWinGUI::SetColor(unsigned long)':
D:\Hobby\Arduino\arduino-1.8.7\hardware\STM32GENERIC_huaweiwx\STM32\libraries\STemWin532\src/STemWin532.h:46: undefined reference to `GUI_SetColor'
sketch\draw_2d.ino.cpp.o: In function `emWinGUI::FillRect(int, int, int, int)':
D:\Hobby\Arduino\arduino-1.8.7\hardware\STM32GENERIC_huaweiwx\STM32\libraries\STemWin532\src/STemWin532.h:113: undefined reference to `GUI_FillRect'
sketch\draw_2d.ino.cpp.o: In function `emWinGUI::SetAlpha(unsigned char)':
D:\Hobby\Arduino\arduino-1.8.7\hardware\STM32GENERIC_huaweiwx\STM32\libraries\STemWin532\src/STemWin532.h:213: undefined reference to `GUI_SetAlpha'
sketch\draw_2d.ino.cpp.o: In function `emWinGUI::SetColor(unsigned long)':
D:\Hobby\Arduino\arduino-1.8.7\hardware\STM32GENERIC_huaweiwx\STM32\libraries\STemWin532\src/STemWin532.h:46: undefined reference to `GUI_SetColor'
sketch\draw_2d.ino.cpp.o: In function `emWinGUI::FillRect(int, int, int, int)':
D:\Hobby\Arduino\arduino-1.8.7\hardware\STM32GENERIC_huaweiwx\STM32\libraries\STemWin532\src/STemWin532.h:113: undefined reference to `GUI_FillRect'
sketch\draw_2d.ino.cpp.o: In function `emWinGUI::SetAlpha(unsigned char)':
D:\Hobby\Arduino\arduino-1.8.7\hardware\STM32GENERIC_huaweiwx\STM32\libraries\STemWin532\src/STemWin532.h:213: undefined reference to `GUI_Font32_1'
D:\Hobby\Arduino\arduino-1.8.7\hardware\STM32GENERIC_huaweiwx\STM32\libraries\STemWin532\src/STemWin532.h:213: undefined reference to `GUI_SetAlpha'
sketch\draw_2d.ino.cpp.o: In function `emWinGUI::SetPenSize(unsigned char)':
D:\Hobby\Arduino\arduino-1.8.7\hardware\STM32GENERIC_huaweiwx\STM32\libraries\STemWin532\src/STemWin532.h:50: undefined reference to `GUI_SetPenSize'
sketch\draw_2d.ino.cpp.o: In function `emWinGUI::SetTextMode(int)':
D:\Hobby\Arduino\arduino-1.8.7\hardware\STM32GENERIC_huaweiwx\STM32\libraries\STemWin532\src/STemWin532.h:166: undefined reference to `GUI_SetTextMode'
sketch\draw_2d.ino.cpp.o: In function `emWinGUI::SetFont(GUI_FONT const*)':
D:\Hobby\Arduino\arduino-1.8.7\hardware\STM32GENERIC_huaweiwx\STM32\libraries\STemWin532\src/STemWin532.h:169: undefined reference to `GUI_SetFont'
sketch\draw_2d.ino.cpp.o: In function `emWinGUI::SetColor(unsigned long)':
D:\Hobby\Arduino\arduino-1.8.7\hardware\STM32GENERIC_huaweiwx\STM32\libraries\STemWin532\src/STemWin532.h:46: undefined reference to `GUI_SetColor'
sketch\draw_2d.ino.cpp.o: In function `emWinGUI::DrawArc(int, int, int, int, int, int)':
D:\Hobby\Arduino\arduino-1.8.7\hardware\STM32GENERIC_huaweiwx\STM32\libraries\STemWin532\src/STemWin532.h:78: undefined reference to `GUI_DrawArc'
sketch\draw_2d.ino.cpp.o: In function `emWinGUI::SetPenSize(unsigned char)':
D:\Hobby\Arduino\arduino-1.8.7\hardware\STM32GENERIC_huaweiwx\STM32\libraries\STemWin532\src/STemWin532.h:50: undefined reference to `GUI_SetPenSize'
sketch\draw_2d.ino.cpp.o: In function `emWinGUI::DrawPoint(int, int)':
D:\Hobby\Arduino\arduino-1.8.7\hardware\STM32GENERIC_huaweiwx\STM32\libraries\STemWin532\src/STemWin532.h:101: undefined reference to `GUI_DrawPoint'
sketch\draw_2d.ino.cpp.o: In function `emWinGUI::SetTextAlign(int)':
D:\Hobby\Arduino\arduino-1.8.7\hardware\STM32GENERIC_huaweiwx\STM32\libraries\STemWin532\src/STemWin532.h:165: undefined reference to `GUI_SetTextAlign'
sketch\draw_2d.ino.cpp.o: In function `emWinGUI::DispStringHCenterAt(char const*, int, int)':
D:\Hobby\Arduino\arduino-1.8.7\hardware\STM32GENERIC_huaweiwx\STM32\libraries\STemWin532\src/STemWin532.h:136: undefined reference to `GUI_DispStringHCenterAt'
sketch\draw_2d.ino.cpp.o: In function `drawarcscale()':
D:\Hobby\Arduino\arduino-1.8.7\hardware\STM32GENERIC_huaweiwx\STM32\libraries\BSP_f429i_discovery\examples\STemWin_ARDUINO\draw_2d/draw_2d.ino:190: undefined reference to `GUI_FontComic18B_ASCII'
[... 79 lines cut ...]
sketch\draw_2d.ino.cpp.o: In function `emWinGUI::SetDrawMode(int)':
D:\Hobby\Arduino\arduino-1.8.7\hardware\STM32GENERIC_huaweiwx\STM32\libraries\STemWin532\src/STemWin532.h:22: undefined reference to `GUI_SetDrawMode'
sketch\draw_2d.ino.cpp.o: In function `emWinGUI::SetColor(unsigned long)':
D:\Hobby\Arduino\arduino-1.8.7\hardware\STM32GENERIC_huaweiwx\STM32\libraries\STemWin532\src/STemWin532.h:46: undefined reference to `GUI_SetColor'
sketch\draw_2d.ino.cpp.o: In function `emWinGUI::MagnifyPolygon(GUI_POINT*, GUI_POINT const*, int, int)':
[... cut ...]
D:\Hobby\Arduino\arduino-1.8.7\hardware\STM32GENERIC_huaweiwx\STM32\libraries\STemWin532\src/STemWin532.h:112: undefined reference to `GUI_FillPolygon'
libraries\BSP_f429i_discovery\STemWin_Config\LCDConf_stm32f429i_disco_MB1075.c.o: In function `HAL_LTDC_LineEventCallback':
D:\Hobby\Arduino\arduino-1.8.7\hardware\STM32GENERIC_huaweiwx\STM32\libraries\BSP_f429i_discovery\src\STemWin_Config/LCDConf_stm32f429i_disco_MB1075.c:367: undefined reference to `GUI_MULTIBUF_ConfirmEx'
collect2.exe: error: ld returned 1 exit status
exit status 1
Error compiling for board Discovery F429ZI(180M).
```

huaweiwx, 2018-11-12:
Using the Stewin library, you need to  link  the external library STemWIN.lib (no OS), or STemWIN_OS (for the  FreeRTOS), so you need to make the appropriate options from the menu extern lib, and for draw_2d.ino  sketch is  STemWIN.

VictorTagayun, 2018-11-12:
Thanks! 
That worked!
Some observations:
display_2d(), bottom LCD have some lines (looking like a dirt).
drawarcscale(), doing nothing, display just becomes brighter for a while.
draw_graph(), some small lines at the bottom of the LCD.

closed 2018-11-12 by VictorTagayun